Xinyuan-LilyGO/T-keyboard #3: Windows 10 & 11 work fine, but challenges connecting to Android

vcchm, 2022-11-02:
Hi ! I compiled the code, pairing & sending keys work well with a PC Windows 10 or 11, but I can't pair with an Android device (Samsung note 20). Here are screenshots.

[image]
[image]

LilyGO, 2022-11-03:
Hello. Is it possible to pair with this [firmware?](https://github.com/Xinyuan-LilyGO/T-keyboard/releases/download/T-KeyboardV1.0.0/T-keyboard__factory_1.0.0.bin)

vcchm, 2022-11-03:
I only can flash an arduino sketch, not a binary but if i test the original firmware provided with a new keyboard, it pairs after a couple of atempts. Can you tell me the precise version of the blekeyboard you are using ? It seems that they did modifications that makes it not working for Android in the latest versions...

LilyGO, 2022-11-03:
In the lib

vcchm, 2022-11-03:
No the latesy version is not working - you compiled the code with a former version but you do not tell which version you used in the github. Can you tell me (and other users) the precise version number you used ? The latest has a bug, and i could not find the working version

LilyGO, 2022-11-04:
The library has been modified. It is not a certain version

closed 2022-11-04 by LilyGO

vcchm, 2022-11-04:
I don't think it has been modified. What happens is that your page should mention the VERSION you are using so that we are sure we use the working version. Do you know what the version is or at least WHEN you compiled the code so that I can use this date's latest version ?