Xinyuan-LilyGO/T-keyboard #7: Unable to compile BLE_Keyboard

Chanete, 2023-04-13:
Hi, I'm trying to compile the BLE_Keyboard  sample.
I followed the instructions (installed latest arduino ide 2.0.4, installed latest usb driver, addded board, selected ESP32C, added the two libs and compile and upload. 

Everything seems to be OK, but the keyboard is now locked (nothing on screen) 
This is my compiler output. 

Thx for your help

`Sketch uses 988408 bytes (75%) of program storage space. Maximum is 1310720 bytes.
Global variables use 57316 bytes (17%) of dynamic memory, leaving 270364 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port COM9
Connecting....
Chip is ESP32-C3 (revision v0.3)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:d5:ae:d8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00108fff...
Compressed 13152 bytes to 9473...
Writing at 0x00000000... (100 %)
Wrote 13152 bytes (9473 compressed) at 0x00000000 in 0.4 seconds (effective 267.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 329.8 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.2 seconds (effective 435.7 kbit/s)...
Hash of data verified.
Compressed 1019904 bytes to 553498...
Writing at 0x00010000... (2 %)
Writing at 0x0001b21c... (5 %)
Writing at 0x00028388... (8 %)
Writing at 0x000352d2... (11 %)
Writing at 0x000403fd... (14 %)
Writing at 0x00046293... (17 %)
Writing at 0x0004cbef... (20 %)
Writing at 0x000540c8... (23 %)
Writing at 0x0005a325... (26 %)
Writing at 0x0006083a... (29 %)
Writing at 0x00067dec... (32 %)
Writing at 0x0006e40f... (35 %)
Writing at 0x00074d16... (38 %)
Writing at 0x0007af53... (41 %)
Writing at 0x00081230... (44 %)
Writing at 0x000886be... (47 %)
Writing at 0x0008ee4a... (50 %)
Writing at 0x0009696a... (52 %)
Writing at 0x0009d20d... (55 %)
Writing at 0x000a34e9... (58 %)
Writing at 0x000aa543... (61 %)
Writing at 0x000b1eab... (64 %)
Writing at 0x000baa52... (67 %)
Writing at 0x000c14c2... (70 %)
Writing at 0x000c7d5e... (73 %)
Writing at 0x000ceaf3... (76 %)
Writing at 0x000d5047... (79 %)
Writing at 0x000db14a... (82 %)
Writing at 0x000e0eda... (85 %)
Writing at 0x000e75b5... (88 %)
Writing at 0x000eec7c... (91 %)
Writing at 0x000f8596... (94 %)
Writing at 0x000fdd96... (97 %)
Writing at 0x00103f19... (100 %)
Wrote 1019904 bytes (553498 compressed) at 0x00010000 in 13.7 seconds (effective 597.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...`

allthemod, 2023-04-14:
I had the same problem.
It's like you can't write to there.
Anyway I just used the download flash tools to flash the firmware bin file into the esp.
If you wanna to customize it idk.
But if you wanna to return it back to the default I can help you.

Chanete, 2023-04-19:
I've found a workaround that works, I'm using platformIO instead of Arduino IDE and now I can compile and modify without problems.

Probably there is some setup in arduino ide config missing in the docs. 

Thx anyway @allthemod  for your offer.

allthemod, 2023-04-19:
Well I didn't know that.
This is also very handy to me so thank you and have a great day @Chanete .

copykatze, 2023-05-05:
I had the same problem and solved it by changing the "flash mode" in the Arduino IDE from "QUI" to "DOUT". Flashing seems to work fine now. Cheers.

allthemod, 2023-05-05:
Thanks and have a great day.

Adarkk, 2024-06-25:
Hey can someone help me to get taht stuff back to work ?