Xinyuan-LilyGO/T-keyboard #8: display initialization problem

copykatze, 2023-06-11:
Sometimes the display does not initialize properly, it will only show some thin lines or a light grey background and it can't be written to by the program. Reseting the device by on/off or the reset button helps sometimes after a few tries, sometimes it seems to need "off-time", but I could not really figure out when it does not work and what gets it back on track. It is really unreliable like this. What might cause this weirdness?

I found that when I enabled core debug level, at `TFT_099.begin();` I get this log:

```
12:51:23.065 -> [   174][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
12:51:23.097 -> [   174][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
12:51:23.097 -> [   177][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
12:51:23.195 -> [   283][E][esp32-hal-spi.c:227] spiAttachMISO(): SPI Does not have default pins on ESP32C3!
```

Could this be the cause of it? But those error messages will also be in the logs when it works. No idea how to debug this any further... :(

Any help appreciated, thanks and cheers!

copykatze, 2023-06-11:
In `TFT_GC9D01N.h`I found

```
#define TFT_MISO  -1
#define TFT_MOSI  10//21
#define TFT_SCLK  20//22
#define TFT_CS    -1//18  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   -1//27
#define TFT_BL    8//4

/*
#define TFT_MISO  -1
#define TFT_MOSI  21
#define TFT_SCLK  22
#define TFT_CS    18  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   27
#define TFT_BL    23//4
*/
```
It seems strange to me that TFT_MISO, TFT_CS and TFT_RST. are all defined -1 (does that mean "undefined?) and I wonder what that commented block of alternative definitions is all about...

copykatze, 2023-06-11:
Another observation: what seems to help is to power it on while connected to usb c power and then pressing the reset button. This method does not help without the device being plugged in.

So maybe the small battery does not provide enough power for the (tiny) display? But on the other side, the grey screen happens also while powering on when plugged in sometimes...

alejovargas78, 2024-01-22:
Hello, I had the same problem, the solution is simple.

Open the library TFT_GC9D01N.cpp and disable the following lines

int TFT_GC9D01N_Class::begin()
{
     pinMode(TFT_BL, OUTPUT);
     pinMode(TFT_SCLK, OUTPUT);
     //pinMode(TFT_MISO, OUTPUT);
     pinMode(TFT_MOSI, OUTPUT);
     pinMode(TFT_DC, OUTPUT);
     //pinMode(TFT_CS, OUTPUT);
     //pinMode(TFT_RST, OUTPUT);

Greetings from Bogota Col.

copykatze, 2024-01-22:
Wow, thank you so much, @alejovargas78 !! I did not get an external display working (yet), but I am happy to hear that it seems to be possible :)

Could you share with me which kind of display you were able to connect and how you wired it exactly?

I am trying with a ili9341 ("2.2 TFT SPI 240*320") display.

My wiring, after editing the library according to your suggestion, is 

Display <-> T-Keyboard

SDO (MISO) <-> 
LEDK <->LED
SCK <-> SCL
SDI (MOSI) <-> SDA
D/C <-> D/C
RESET <->
CS <->
GND <-> GND
VCC <-> +3.3V

Running the [TFT_Demo.ino](https://github.com/Xinyuan-LilyGO/T-keyboard/blob/main/lib/T-0.99TFT/examples/TFT_Demo/TFT_Demo.ino) example, the display just  flickers a bit.

One error I see on the serial monitor is:
`[   192][E][esp32-hal-spi.c:227] spiAttachMISO(): SPI Does not have default pins on ESP32C3!`

It would be so awesome if I could finish this project afterall :)

copykatze, 2024-01-23:
Unfortunately, I have the same result/problem with the adjustments in the library, still thanks for the suggestion!

alejovargas78, 2024-09-07:
pinMode(TFT_MOSI, OUTPUT);
    pinMode(TFT_DC, OUTPUT);
    pinMode(TFT_BL, OUTPUT);
    pinMode(TFT_SCLK, OUTPUT);
    ledcSetup(1, 1000, 10);
    ledcAttachPin(TFT_BL, 1);
    ledcWrite(1, 0);
    delay(100);

    SPI.begin(TFT_SCLK, TFT_MISO, TFT_MOSI, TFT_CS);

    int i = 2;
    while (i--) {
        lcd_init();
        delay(200); /* code */
    }
    ledcWrite(1, 220);//开启背光

    return 1;