Xinyuan-LilyGO/T-keyboard #10: working like a Blackberry Keyboard

Flexmaen, 2023-11-28:
Hi!

Great hardware, however for me it's a bit confusing that this keyboard does look like a Blackberry-Keyboard, but doesn't work like one.

This means:
- ALT-Key usually does all the other signs shown on the keyboard, not sym.
Numbers, ,?/().,' etc. is all done by alt key.
- Alt-Key works as long as you hold it down, that means just jold it down and type numbers instead having to press it every time you write a number.
- Sym key can be configured for international options. E.g. sym+u = ü.
- right shift key can be configured as normal shift key or as ctrl-key which means that witch ctrl+c and v you can copy and paste things. (maybe it also makes sense to use the sym-key here instead).

slayinturds, 2023-12-21:
So it does work like a blackberry keyboard or not exactly? It's been ages since I had my bold 5000, and I'm just trying to have that magic back via Bluetooth. Is this the device for the job?

Flexmaen, 2023-12-22:
It doesn't even come close to a Blackberry keyboard although it is a keyboard of the Q10.
But the way it works is completely messed up.
Like I said: Alt and Sym Key are confused, holding down the alt/sym-Key has no effect, so typing numbers is a pain.

So if you ever used a Blackberry keyboard, then this device is a big disappointment. Depending on what you need this for, you might take a look on the BBQ20KBD or Help2Type.

However, the good thing: Most things could be fixed via software, that's why I opened this issue here.

slayinturds, 2023-12-24:
That makes sense, thanks for the suggestion. Hopefully they figure out the software since I already ordered one

zenkalia, 2023-12-27:
I just bought one because fixing some keybinds seems easier than taking the BBQ and connecting it to a battery and BLE module for my use case (typing with baby asleep on me).

I'll be taking a look at this when mine comes. At a glance, the right shift seems easiest to fix. Other language support seems daunting and the other modifiers seem somewhere in between. Thankfully the shift issue is the most important to me!

slayinturds, 2023-12-27:
I hate it. I'll be taking it back and if they refuse I'll go to their
office in Gainesville FL. I live there conincidentally. If it's possible
I'll share feeeback directly. The keys for one are at the same level as the
case and the trim pieces, making it so difficult to land on the keys and
not the case elements when inputting characters. That is such a no brainer
to fix and figure out when prototyping

zenkalia, 2023-12-28:
That's too bad! Mine hasn't come yet and the ergonomics of those bars does worry me. This is the best option out there since the Help2Type is so expensive that you might as well buy a palmtop computer.

slayinturds, 2023-12-28:
Yeah it is.
Honestly, I'm interested in this one next that you suggested:
https://shop.pimoroni.com/products/bbq20-keyboard-with-trackpad-usb-i2c-pmod?variant=39913457188947

Flexmaen, 2023-12-29:
What are you planning to use it for? I didn't test the BBQ20, but I'm currently using the Help2Type for my phone. The flip-mechanism works better than I thought it will. The downside compared to a Blackberry keyboard is that keys are harder to press, and the lack of the ctrl-key. The layout is a bit different and some things like the delete key need a bit to adjust. But otherwise the layout is in full accordance with the virtual keyboards that gboard or iOS provide, so it's not a stupid change in the layout. All in all, for me it was worth buying it.

I'm using the T-Keyboard for my Fire-TV now. For occasional typing it's better than the remote control, but that's it.
Sure this is a rather cheap solution, but now I'm not the only one who is surprised how bad a Blackberry keyboard can be if you get the software and the case wront.

slayinturds, 2023-12-29:
With the phone propped up by the kickstand ring on the back of the phone,
and the keypad detached in my hand, I thought I could type emails better
when away from my computer. That may not be the case. I used to be able to
type for a long time for emails with my BB5000, but I'm probably chasing a
time that is irrelevant. I was thinking about your situation and not
wanting to make sound the other day, because I just purchased a Logitech
wireless mouse that makes no clicking sound. I thought I'd hate it but
actually I love it; the M650. I see your uses, that makes sense. The
Help2type, looks ergonomic but I could not deal with the hard key pressing.
I'm so picky with this stuff, I don't mind searching for the perfect
solution to all my problems.

zenkalia, 2024-01-05:
Just opened my first PR to this end: https://github.com/Xinyuan-LilyGO/T-keyboard/pull/11

Next comes swapping `alt` and `sym` and adding arrow keys. I'm thinking sym + WASD because it's just too painful not having arrow keys.

And a note on ergonomics: I don't mind the bars between the rows but the case itself IS a little too tall. It should have had a bevel or something because as it is, the keys that are near the case are a little difficult to press.

zenkalia, 2024-01-05:
Alright I got a little overzealous last night and finished it. ALT and SYM swapped and WASD arrow keys:

https://github.com/zenkalia/T-keyboard/tree/alt_sym_arrows

This is what I'm running on my T-Keyboard right now and it's much more fluid than the default behavior. 👍

ibrand, 2024-01-05:
@zenkalia very cool, I just got my T-keyboard in the mail, started searching around for this and was lucky enough to find your comment from just a few hours ago. If I were to use your fork, would I follow the Arduino IDE setup flow and then import your example files? How did you set your keyboard up with alternative code?

thanks for your work on this 🎉

zenkalia, 2024-01-06:
I used the vs code instructions. The arduino IDE was way less fleshed out from what I could tell.

There's also a necessary driver for pushing to the serial port of the device (chx something) which was terrifying to install but it worked. Would love to have that come with the PlayformIO board config or have LILYGO bless it in the readme.

zenkalia, 2024-01-06:
Yeah I've been thinking about how to best handle custom keys since they're different for everyone. I think the right move is to have an extra mapping matrix for symbols that people can modify in code. If you leave a given cell empty then it can be a fully custom function (see backlight toggle).

If there are ever enough users to justify menus and stuff then we'll need to change from the 8x20 font to an 8x10 so that we can display 4 rows rather than 2... Also need a better way to render stuff to the screen since the TFT library isn't going to give us scrolling, for example.

I plan on adding the symbol mapping matrix later today since it's a more elegant and extensible way to add arrow keys.

After that on my todo list is using both rows on the TFT screen when typing.

Also the lack of key repeat is driving me nuts!

zenkalia, 2024-01-14:
@Flexmaen So I did put in a mapping table for the sym key until I realized that those other special characters don't exist in the font. Also I have no idea how this font works to add them. 🤔

zenkalia, 2024-01-14:
Ok. I know how the font works. It's kind of bonkers.

So the font file is a bunch of bitmaps. Here's A: https://github.com/Xinyuan-LilyGO/T-keyboard/blob/main/lib/T-0.99TFT/src/Fonts.h#L38-L39 each hex pair is a row in the character. Each bit is a pixel being on or off.

We write text to the screen using `TFT_099.DispStr`. Each character in that function is then converted from ascii value to "ord" like we see in that font file. The helper function that does that is here: https://github.com/Xinyuan-LilyGO/T-keyboard/blob/main/lib/T-0.99TFT/src/TFT_GC9D01N.cpp#L533-L550 .

So, for every character that you want to add, you have to add it to this conversion helper and to the font bitmask. How much work would it be to make this font array indexed by ascii value instead of "ord" value? 🤯